LabVIEW-DCAF/TagEditorCore #133: PC Editor Node Should Not Require Valid Configuration File Path

Beazurt, 2015-01-17:
As far as I know the Configuration File Path is only really needed for the deploy tool to work properly.  I don't think we should return an error if this field is empty.  If they go to deploy something and it doesn't work they can always come back to this setting.  We might want to turn this check off for the cRIO as well (unless I am missing a use-case).

Beazurt, 2015-01-17:
@dsmithni I changed this to report as a warning.  You can still see the error message in the 'View Node Data', but not specifying this doesn't prevent your application from running anymore (no more red X).  Want to make sure there wasn't a use case I am missing before closing this.

dsmithni, 2015-01-18:
seems fair. thats why I was careful to make the viewer also show warnings

closed 2015-01-18 by dsmithni