LabVIEW-DCAF/TagEditorCore #134: Includes File Path on Editor Node Should Default to Relative Path of Config File

Beazurt, 2015-01-17:
I'd like to specify the module includes file path for the shipping example, but the path will almost certainly be different for each user.  I think this path should default to be relative to the .pcfg file path so that it can move across machines.  We could also allow users to specify an absolute path but as a secondary feature.

becega, 2016-07-20:
I think it is. This issue is related to #226. I am closing this one as we are working in the other one

Beazurt, 2016-08-11:
This looks like a different issue than #226 to me. The path to the includes VI under a target in the editor is absolute. If the code moves on disk, or if the configuration file is opened up on another development machine, the path to the includes VI is no longer valid. 

Most development projects will likely move the source code and configuration file together. If we change this path to be relative (or at least have the option to be relative) to the path of the config, then things can move around without issue.