LabVIEW-DCAF/TagEditorCore #135: Moving an Engine That Contains a Module From One Target to A Different One Breaks the Engine

Beazurt, 2015-01-17:
Add two targets to a system and move an engine containing a module from one target to the other.  This puts a red X on that engine and returns the following error...
538303
Source:
validate standard engine configuration
<APPEND>Internal Error: Module "BCB384FB-D15F-4DB1-B3E9-72DBF8C48669" is registered but does not exist in the parent class

Beazurt, 2015-01-17:
@dsmithni This looks like it might have been as simple as backwards boolean logic in the Std. Engine 'Validate' method, but I wasn't 100% sure what you were checking for so I haven't changed it.

dsmithni, 2015-01-18:
Lol booleans. all thats supposed to do is make sure there is no mismatch between the child and parent, since the parent has a list of modules and so does the child. I couldn't think of any reason this wouldnt work yesterday but booleans would def explain it

dsmithni, 2015-01-20:
fixed in refactor engine branch, doesn't belong here

closed 2015-01-20 by dsmithni