LabVIEW-DCAF/TagEditorCore #146: Change name of package to Tag Bus Data framework

Beazurt, 2015-02-16:
The product name is currently 'TBM Tag Editor Framework'. We should change it to 'Tag Bus Data Framework' before publishing 2.0.

dsmithni, 2015-02-18:
no

dsmithni, 2015-02-18:
That name change was purposeful. First it prefixes everything so if you type TBM into VIPM you find everything. Second it is the editor portion of the framework, not the whole framework. The whole framework is the VIPC file. The editor portion was named that way to indicate that we also plan on using it for TCE if we ever update that.

Beazurt, 2015-02-18:
OK, prefixes are helpful, but I don't think having a bunch of packages which don't include the name of the framework anywhere is a good idea. I personally think two prefixes is better TBM for the modules TBD for everything else. If we want to have a single prefix I think using TBD would make a lot more sense.

dsmithni, 2015-02-18:
No, dont change the prefix. Just keep it. Nobody will ever notice or care besides you. Our current customer was very happy with the prefix and thought it was extremely nice to be able to find everything in one place.

Also, the framework is not one package so its more confusing to claim that it is. Imagine if I told you "install tag bus framework plz" and you were new to the project. You'd install the package called "tag bus framework". In reality what we want them to install is packages W X Y and Z. Makes more sense for the VIPC package to have the "framework" title while the subpackages are just subpackages. Way less confusing.

Beazurt, 2015-02-18:
This isn't a big deal so I'm ok with leaving it alone if you disagree. I don't understand the resistance to the change though so I want to make sure we fully understand each other.

It seems there are two points of disagreement. One is on the overall name of this package and the other is on the use of prefixes. I think the main reason I don't like the name of this package is that so far we haven't talked about a Tag Editor Framework anywhere in our documentation or in the code. As far as I know this is the only place we've ever used those three words together. Of course we could start adding the phrase places, but I'm not sure the new concept adds much value. Also while I agree that the framework is really a collection of packages and that this one primarily installs editor code, this is currently the top-level required package, and all of its subpackages are listed in its dependencies. If we told someone to install TBD framework, and they installed this with its dependencies, they would have the whole framework installed. Also I disagree that no one will ever care about the name of this. Everyone that I have had install packages so far has tried to have at least a basic understanding of what each one is for.

The prefixes are a great idea. Especially once these get posted to the Tools Network, we will want people to have an easy way to find everything. I have two concerns with using TBM as the only prefix though. First it doesn't distinguish between a dependency of the framework and a framework plug-in or extension. We could very easily get to a point where the majority of TBM-prefixed packages aren't actually needed. Of course the downside of two prefixes is that its harder to remember, but again installing TBD should automatically install its dependencies so I don't think that's a big issue. Second if we want one prefix for everything, I don't understand why we would rather have people search using TBM instead of TBD.

dsmithni, 2015-02-18:
I'd rather people search for TBM instead of TBD because its already done. Other than that I don't care. Just like the install problem we've had in the past nobody cares if they're installing stuff they don't need. It takes less time to install an unneeded package than it takes to think about what packages are needed. I do that all the time with openG. When I see one of the files missing I say "oh dammit now I have to install all that openG stuff again" and install everything indiscriminately.

Tag editor core is not the whole framework although the dependencies would as you say install all the necessary components. It wouldn't  however install the sample projects or anything that makes it useful (how many times have you opened up the editor only to remember crap I forgot to install the standard engine -- I've done it many times).

Finally, "tag editor core" is...well, the core of the tag editor. I don't think it needs much more explanation, although I may be wrong.

Beazurt, 2015-02-18:
So I am totally fine with 'Tag Editor Core' instead of 'Tag Bus Data framework', because 'Core' at least gives the right connotations for what the package is. However the package is currently called 'TBM Tag Editor Framework' which I think is confusing. It suggests the existence of another framework that we haven't explained anywhere. We also define TBM as a plugin to the framework which this isn't.

Most of the existing TBM prefixes wouldn't need to change if we also added a TBD prefix. People could also still find both TBM and TBD prefixes by searching for just TB. Also both prefixes show up right next to each other in VIPM so the two prefixes wouldn't force people to go hunt through the long list of packages. There is also a very clear line between the two prefixes with the exception of the Standard Engine.

dsmithni, 2015-02-18:
Oh I see, yes I agree it should be tag editor core like the repo, rather than framework. TBM Tag Editor Core or whatever.

The problem still lies with the fact that most people will want some of the modules. Its far easier to explain the package meanings to the one discerning individual who wants to select specific modules than to explain to everyone why some stuff has one prefix and some stuff has another.

Beazurt, 2015-02-18:
Glad the package name misunderstanding is cleared up.  I'll go ahead and change the name to 'Tag Editor Core'. 

I still don't think I fully understand the prefix use-cases that you have in mind so I want to pry just a little further (sorry I am sure this is annoying but I usually see things eye-to-eye with you and I still don't here).  

This is where I am coming from, let me know what I am missing. Most users will install the framework from a VIPC file that we publish somewhere. That VIPC file will include some of the most helpful TBMs, but not all of them. If users want to find more TBMs, they type TBM in the search bar and find a list that they can install. I think many people actually will selectively choose them, especially if we continue to make a lot more, but if they want to install all of them that isn't hard to do.

In some cases, people may not start with the VIPC file and will want to install the packages directly from VIPM. In that case they will likely search for TBD in package manager and wouldn't know to use the term TBM. Some will also likely want to know what packages are required to install and which aren't. Once they install all of the TBD packages, the TBD examples package will have a dependency on some TBMs, so they will have some TBM prefixed modules in their package manager already. Once they see both prefixes, I don't think the TBM/TBD convention will be hard to explain, I think most people will just get it.

There is very likely a good reason this is a bad idea, but I think the distinction will help some people and still don't understand how anyone would be negatively affected by it.

Beazurt, 2015-02-18:
Also to be clear, here are a few more examples of why I think the prefix matters. We have code for sample projects in a package called 'TBM Development Kit' which by name seems like its only for TBMs, but it also has TBD and Editor sample projects in it.

Also when building the TBD Examples package, I wanted to name the package TBD Examples. With this prefix convention it would either be TBM Examples or TBM TBD Examples.

becega, 2016-03-23:
Should we now change the prefix to DCAF?

Beazurt, 2016-03-23:
I think the TBM prefix is still fine for the module plugins. I'd rather use DCAF than TBM for everything else though.

becega, 2016-07-20:
It is DCAF now

closed 2016-07-20 by becega