LabVIEW-DCAF/TagEditorCore #162: Editor Allows Duplicate of Modules that Have Met Their Maximum Instance Count

Beazurt, 2015-03-10:
This might be hard to fix for the trouble it causes, but a module like the scan engine only allows one instance per target.  Once someone adds one instance, we remove it from the list of modules that can be added to that engine.  However, someone can still right-click the scan engine module and select 'Duplicate'.  We will then add a second instance.  If its too hard to fix, we could instead add a check to the target's config valid that verifies this and warns a user where the problem is.

becega, 2015-03-10:
If there can only be an instance of a module in an engine should we allow it to be duplicated at all? The easiest would be disable duplicate if the number of instance allowed is 1.

dsmithni, 2015-03-10:
module determines if it can be duplicated so we'd either have to traverse the tree or copy some of the engine's code. Not hard but annoying. May be able to add to the TBM parent.

dsmithni, 2015-03-10:
do we want to enable duplication at all?

dsmithni, 2015-03-10:
We can fix it for the case of 1 but I think anything we try to do to fix the general case will lead to deadlock. We need to get siblings which CEF doesn't give us, so we'd need to get all of the parent's children except self, and self is locked, which means we're very likely to deadlock. May be able to fix this using variant functions to find the classname but...lets table this for now

Beazurt, 2015-03-10:
Yeah, this isn't a big deal.  I'm not sure how helpful it is to duplicate a module.  Doing so for an engine or a target seems a lot more valuable.  If we do want to fix it maybe its best just to turn Duplicate off for all modules?

dsmithni, 2015-03-10:
I think so, since it probably breaks things if we don't add everything just right.

This whole managing state across different objects thing is super annoying.

becega, 2016-12-05:
WE need to change the modules to decide itself if it can be duplicated. The scope of the modules was changes to target and there is a array of the classes in the target that contains all modules configuration classes names. With this we can count if there are enough when duplicating a module. so there is no need to transverse the channel. Closing this and open several in the respective modules

closed 2016-12-05 by becega