LabVIEW-DCAF/TagEditorCore #180: Auto-add engine on target creation if only one engine type defined

pollockm, 2016-02-12:
When creating a target in the configuration editor UI, it is necessary to first create a target (such as cRIO), then to create an engine on that target.  This step seems redundant if there is only one engine type available for the user to choose from, and we know that the user must add an engine to further configure the target.

Can we auto-add the engine to a newly created target if only one engine type is possible?  I expect this will be the standard experience for a new (and probably most existing) framework users.

Beazurt, 2016-02-16:
I think we may have multiple engines or even other types of services to add under a Target in the near future. The auto-add logic isn't a bad idea, but it may not get used for long. There may also be cases where it is confusing if the intended plugins aren't installed. Could creating template configurations be a good alternative for solving this problem? Users could go to File>>New and see multiple options instead of just getting a blank configuration.

pollockm, 2017-01-31:
@Beazurt - I haven't seen much in the way of other engines/services happen in the last year.  I'd like to get this feature into the 1.2 release, just as a developer convenience.