LabVIEW-DCAF/TagEditorCore #189: Deploy tool: handle timeout error

pollockm, 2016-02-12:
Attempting to deploy to a target with an incorrect IP address results in a long (~15 second) timeout followed by an unhandled error 363500 at an unidentified location.

[image]

Instead, I'd like a nicely handled error, or at least a custom message.

The timeout is rather long, might want to shorten it as well.

becega, 2017-01-31:
Might be fixed with latest improvement to the tool

pollockm, 2017-01-31:
The message is actually less verbose than before (we return the command line error output from pscp).  Apparently the timeout for TCP retries used by pscp comes from the system registry, not a command-line settable input, so this is unlikely to get truly fixed any time soon.

jwgorman, 2017-04-13:
Acceptance Criteria
- A more helpful error is displayed

theSloopJohnB, 2017-04-20:
This will be in an upcoming PR, but I want to note it here in case I forget. The error you will get is:

> Verify that you have specified the correct IP Address for your target and that your target is on the network

closed 2017-05-03 by theSloopJohnB