LabVIEW-DCAF/TagEditorCore #198: Create scale parent UI

pollockm, 2016-04-13:


pollockm, 2016-04-13:
Current thoughts here:
[image]

https://dcaf.mybalsamiq.com/projects/scalingui/hybrid

Beazurt, 2016-04-14:
This mockup is outstanding. Could we make more room on the UI by hiding the 'Configuration' pane somewhere? My feeling is that a user wouldn't interact with this often after the first definition of the scale. Maybe have a Mappings, and Configuration tab so that you can only see one at a time?

Also as a feature request, I think it'd be useful to have an export/import button for a list of scales. This would allow scales to be reused across various engines. The mappings would not be exported/imported, just the number of scales and their configuration.

Beazurt, 2016-04-14:
BTW I don't think we need the Import/Export feature in the first version. I just brought it up to get your thoughts on how something like that could fit into the mockups.

pollockm, 2016-04-14:
Yes, I currently have the mappings and configuration sections as a two-page tab control, since I ran out of space when implementing in LV.

Re: Import/Export: Not enough time to get this done for first release, but nothing preventing it for later.  Add it as an issue to the new Scaling repository and I'll add it to the backlog.

pollockm, 2016-05-23:
Ok, scale parent looks solid at this point.  Closing.  https://github.com/LabVIEW-DCAF/Scaling

closed 2016-05-23 by pollockm