LabVIEW-DCAF/TagEditorCore #199: Add Categories to nodes to add to engine

becega, 2016-04-14:
Add Categories to nodes to add to engine
With the number of available nodes getting larger, the list of nodes to add to an engine will soon be hard to read. Need to add categories to the nodes so they are shown under a folder.

Beazurt, 2016-04-14:
We need to agree on the names of the categories that we want to use. We should limit the number of categories and their overlap, while providing a mechanism for users to define their own. Here are some ideas for brainstorming, but I'm not super happy with them yet.

Categories:
Channel I/O - Scan Engine, FPGA Interface, DAQmx
Protocol I/O - Modbus, Profibus, etc.
Data Processing - PID, Most User Control Modules
Data Services - TDMS, 

I'm not sure how to categorize these modules though. Maybe if we can figure these out the rest will be easier.
Where do we put other I/O like CVT, Web Services, and UI? Is this a different I/O category or do they fit into the existing ones? If we just have one I/O category I fear it will be too big, but maybe that is OK.
What is Alarming? Is it processing or a service?
What about things like benchmarking, Watchdogs, and Diagnostics? Can these just go in an 'Other' category?

Ideally we can get to a flow where people add their I/O, then add their processing, then add their services/other and deploy.

pollockm, 2016-04-14:
What about things like benchmarking, Watchdogs, and Diagnostics - Utilities

Local IO (Scan Engine, CVT, FPGA, DAQmx, etc)
Protocol IO (Modbus, Web Services, etc)
Processing - Alarming
File IO
Utilities (Watchdog, benchmark, etc)

becega, 2016-04-14:
Looking a bit into this, the changes we need to do is Overide Shortcut Menu in the Engine class. Also need to add a new field to the class cache for modules. And Add the extra fields to the editor node. It would be an enum with the defined categories and a string, that will be selected with other. 
Also update all Modules to use this. 
What would be the expected behavior for modules that have this empty? just put them as other? Or custom?

Beazurt, 2016-04-14:
Good question. I think we put them in 'Other'. We can use a string for a custom category, like 'My Company Name' modules, but if that is blank then we just put them into 'Other'.

Beazurt, 2016-04-14:
I like 'Local IO' and 'Protocol IO' a lot. Maybe we call 'Protocol IO' 'Remote IO' instead though so it's a more obvious subset? Something should only ever be local or remote, but not all remote I/O is necessarily a protocol.

I also like Processing.

Could we add File I/O to Utilities? Right now we only have the TDMS logging as a type of File I/O. If we remove File IO and add an 'Other' category then we have 5 categories. I think it'd be good to avoid having much more than 5.

tannerblair, 2016-04-18:
Any thoughts on allowing the end user to add categories, such as "Favorites"? I also think there's some merit in allowing a module to belong to more than one category.

pollockm, 2016-04-18:
I see a feature like that as valuable once we get so many modules(even in the categories) that users can’t easily find what they need.  I think that this can be safely deferred for a later time, unless the feature is trivial to implement and imposes no new burdens on module developers.

----
Matt Pollock
Senior Systems Engineer, Embedded Systems
(512) 683-9435 | matthew.pollock@ni.com

tannerblair, 2016-04-18:
I feel confident in saying that multiple categories would be trivial, and custom categories less so but still very doable. I'll start with multiple categories and go from there.

tannerblair, 2016-04-25:
Categories can now be added by overriding the Get Category method for your module. I didn't create fixed categories, but this would be trivial to add in the future if it seems necessary.

closed 2016-04-25 by tannerblair