LabVIEW-DCAF/TagEditorCore #200: Refactor 'rename channel.vi' template

pollockm, 2016-04-15:
Current template first checks for existence of old tag name, then new tag name, then actually does the rename.  Recommend that we use 'get channels' function to get the channel array used for these checks rather than assuming that the configuration is just a channel array.  One less place a user will need to make edits if they have a more interesting config.

pollockm, 2016-04-15:
Perhaps do the error checks (for whether old name and new name tags exist) in the parent call and core out the rename functionality itself to the user modules, with the assumption that checks have been passed.  Less code overall for folks to write, and less opportunity to make a mistake.

becega, 2016-07-20:
This should be fixed and replaced with Lines.

closed 2016-07-20 by becega