LabVIEW-DCAF/TagEditorCore #234: Allowable instances of module has engine scope - should it have target scope?

pollockm, 2016-08-08:
Example:  LED module.  Each execution target only has one LED to blink.

I can only add one LED module to an engine on a cRIO (because the max instances is 1).  If I add a second engine on the same target, however, I can add LED to each.  Bad.

Beazurt, 2016-08-11:
I'm 95% certain that target-scope is how this field was originally intended to be used. We just never bubbled this logic all the way up to the target within the editor. I am having a hard time imaging times where it isn't OK to have two instances of a module within an engine, but it is OK to have two instances on the target.

smithed, 2016-08-12:
I don't think we ever thought about it (I think the N engines concept came up after the M modules concept was settled) but I agree, that makes the most sense.

becega, 2016-11-23:
I agree it shoudl be Engine scope. Still this gets a little bit trickier, as we can not really go through the tree from inside the node as it will deadlock. So we would either include in the engine a count of modules. OR count the in a FGV. Or add to the engine a way to get all children except the one making the call.

becega, 2016-11-23:
The discussion in #162 is relevant to this one

commit 74540f87 (2016-11-24): Fixes #234
Many VIs change because of the class updates. But the main changes are:
Targets class. Added an array to store the names of the configurations in the target. Added the Data access methods for this array.
Engine Parent Node: Edit Options, AddChildToNode, Remove child to node.
Mass compiled for issues on build
Removed the To and From Repo test as they are now in CEF
closed 2016-11-24 by becega