LabVIEW-DCAF/TagEditorCore #238: Error 1556 while configuring targets

becega, 2016-08-11:
Error 1556 occurred at To More Specific Class in PL Framework Configuration Tool.lvlib:module editor node.lvclass:locate engine.vi->Standard Editor.lvlib:Refresh All Nodes.vi->Standard Editor.lvlib:Configuration Utility.vi

The error happens when you edit a target and there are no engines under that target. If you add a engine the error goes away

closed 2016-08-11 by pollockm

ecleary42, 2019-03-11:
I have a Veristand target running on a Real Time OS on a PC, I am getting this error code on boot up however I have checked that we have a Veritand target on the PC and we do, reinstalled all software on the target and still get this error.  Can still deploy projects to the engine and it runs along side the host with no issues, we are trying to talk to 2 cRio's using custom devices which isnt working, had the custom devices running using a Windows version of the project running the engine on the host over the same network we have the RTOS target running so all should be good for networking.  Would like to clear this error and see if it is whats causing the RTOS from talking to the custom devices or is it something other than that.

Any thoughts or suggestions would be appreciated

pollockm, 2019-03-11:
Hi @ecleary42 - are you using DCAF, Veristand, or both?  If you are using Veristand, but not DCAF, then you probably ended up on this page from a search for that error code, which is a very common error code when working with object-oriented frameworks in LabVIEW.

ecleary42, 2019-03-11:
Just Versitand, apologies if I have posted to the wrong page 

Came from another post with the same issues on a cRio and the answer to that issue was flagged as already resolved on this page

pollockm, 2019-03-11:
@ecleary42 Yep, this page is just related to the DCAF framework, so we won't be the right folks to help you.  Please try the NI forums for more assistance.

ecleary42, 2019-03-11:
done :) 

thanks for the replies and letting me know

commit 08c012bb (2016-08-11): Bug fix

Fixes #238