LabVIEW-DCAF/TagEditorCore #242: Control module error classifications are reported wrong

pweibring, 2016-08-25:
Errors classifications defined under the a control modules "Error Handling" tab gets the wrong classification when reported. It looks like "error classifications" and "classification codes" are erratic already in the parent "control module runtime.lvclass classify error", but there might be issues on other levels as well.

pollockm, 2016-08-26:
@pweibring : I'm looking at error handling in the control module template today.  I don't see an immediate bug in the classify error VI - can you tell me what steps you followed to see an issue?  I'll work on reproducing today if you can tell me how you saw the behavior.

pollockm, 2016-08-26:
Specifically, how are you measuring how they are reported?  Fault list?

pweibring, 2016-08-26:
I added classifications (trivial, critical etc) to the error handling tab of a control module in the editor

I added an error cluster control to the control module runtime user process of my module so I can generate an error when wanted (single or continuous)

I run the system and apply different error codes and the watch engine status faults reports back

I also look at the user process module input cluster to see if the engine clear the error or not for the classification I specified 

I then restart the RT targe

pollockm, 2016-08-26:
The classify error function in the control module runtime passes manual testing.  I'll look downstream for issues in the engine error handling.

gcodetechnology, 2019-09-04:
I believe the issue I am seeing is somewhat related. To test error reporting, I generated an error in one of the override methods of the runtime class of a dynamic plugin. I made sure all errors for this plugin were classified as "unknown" in "classify error.vi". The error gets reported as critical. So it seems there is no possibility of classifying errors. I believe this makes the Error Actions settings in the Standard Engine Module Settings tab irrelevant.