LabVIEW-DCAF/TagEditorCore #245: Deploy Tool Non-functional for CompactRIO

leesecho, 2016-09-21:
Receiving Error when trying to deploy the configuration file or syslog files to a cRIO with NI Auth login credentials:

Error 363516 - LabVIEW: Client does not have access to the specified resource (access is forbidden)

smithed, 2016-09-22:
You should probably add what hardware (linux rt or vxworks) and what specific paths its trying to deploy to.

leesecho, 2016-09-22:
Good point Daniel. Thanks.

And the winners are...
- Linux RT target (cRIO-9068)
- /home/lvuser/natinst/bin/<filename>.pcfg

smithed, 2016-09-22:
Oh too bad those should both work. Did you mean to close the issue btw?

leesecho, 2016-09-22:
No, I didn't! It's like, you know me:)

Beazurt, 2016-10-12:
Is this issue specifically caused by the NI-Auth credentials being required? The tool does work if a cRIO doesn't have them enabled correct? If so this is expected behavior. It'd be good to change the title to include NI-Auth and change the label to feature instead of bug.

leesecho, 2016-10-12:
I traced the root cause of this error to a missing NI Syslog Library dependency of DCAF Tag Editor. The issue has been renamed.

becega, 2016-11-23:
Does that still happen with all the latest versions?