LabVIEW-DCAF/TagEditorCore #265: instance count allowed should require override

knoxberry, 2017-01-12:
If instance count allowed isn't overridden, the editor module is added to the ignored class list without warning. This VI should require the decedent class to override.

Beazurt, 2017-01-12:
I haven't looked at the code again yet, but I thought that if there was no
child implementation then the parent implementation would be called and
would return -1. That would mean that the override is not required. We can
definitely take a look and see what is happening here though.

knoxberry, 2017-01-13:
Yeah, it looks like the parent class is meant to be overriden, so it throws an error if the child class doesn't have a specific implementation. Should be pretty easy to fix. I'm guessing this method should return -1 if no override like you mentioned.

Beazurt, 2017-01-14:
Yep. Confirmed this is a bug. Will try to fix in the next release.

pollockm, 2017-02-16:
@Beazurt - should this be fixed by setting the parent to not throw an error if it got called, or by setting the parent method as 'must override'?

Beazurt, 2017-02-16:
I think making this 'must override' is excessive. I think having the parent return -1 by default (and documenting this in the VI documentation) would be fine in this case, and then users can choose to override if they need non-default behavior. There may be better alternatives though.

pollockm, 2017-02-16:
...the root implementation of instance count allowed is defined in the CEF package (NISE_CEF_serializable configuration.lvclass).  We can provide an override in the TBM Module configuration class that provides the -1 value as default.  That should take care of the issue for DCAF without changing CEF behavior.

[image]

Beazurt, 2017-02-16:
Makes perfect sense to me.

closed 2017-02-16 by pollockm

smithed, 2017-02-17:
I would suggest using 1 instead of -1. 
If a developer creates a module with a true max instance count of 1, but forgets to override if default = -1
-Its tough to catch, unless they think about the fact that their module only supports 1 instance and also deliberately test for that case
-Failing to set the correct value can lead to an error condition if deployed
If a developer creates a module with no max instance count but forgets to override if default = 1:
-A user (or the developer) is temporarily inconvenienced and makes a bug report

Beazurt, 2017-02-17:
I agree that using 1 would be the safer and more conservative way to go, but its hard for me to come up with scenarios where anything too bad would happen if more than one instance of a module was added to a config. that shouldn't allow it, other than errors during initialization. In that case, an end user can fix the problem on their own by changing the config. and not have to change the code (unlike having to change a 1 to a -1 to run more instances). I also think -1 is probably the more common value and would require a lot fewer overrides. There are tradeoffs, but I still think -1 has the upper hand. I'm curious to know if there are other opinions. Its worth a bit more discussion because once we release a value here it would be really bad to change it.