LabVIEW-DCAF/TagEditorCore #269: Editor class has Runtime class dependencies

leesecho, 2017-01-27:
Callback Tag.vi - typedef constant present which is part of Runtime class
Get Object Refs from Node.vi - Something present in the disabled case (Create Test Configuration.vi?) has a dependency on Runtime class

becega, 2017-01-30:
The CallbackTag.VI got fixed with  [364fb86](https://github.com/LabVIEW-DCAF/TagEditorCore/commit/364fb86ae54e3b3ec1d6b68596a4ebfdcbfa91d9) Deleted orphaned constant

THe GEt Objects Refs node has an issue because it is calling the Create Test Configuration VI, which is part of the Test Library which also runs test on runtime and that generate the dependency. We need to move that VI to a new library

becega, 2017-01-30:
FIxed with https://github.com/LabVIEW-DCAF/DCAF-Project-Templates/commit/671e44993bec5e4eec786033822795881b49cf8e bymaking a change on the project

closed 2017-01-30 by becega