LabVIEW-DCAF/TagEditorCore #281: Deploy tool fails if SSHD isnt enabled on CRIO targets

brbrindle, 2017-02-15:
New method to deploy config works only when SSHD is enabled on CRIO targets.
Error message from Putty say port connection actively refused. 
Better error message and/or fall back to old method for directories inside /home/lvuser/

pollockm, 2017-02-15:
Error message - no problem, makes sense to intercept that error and provide a more useful message.

Challenging to fall back to old method - we need files to go in other directories besides /home/lvuser, and WebDAV won't do that (even if using credentials that have permissions in those locations).   Requiring SSH was the best compromise we could find.

Also has the benefit that a user can easily turn off the ability to change config by disabling SSH.

pollockm, 2017-02-16:
Is it correct to believe that the error message that you saw was "Fatal: Network error: Connection refused" ?  I'll put in an error trap for that message so we get a more useful message (something like "Connection refused.  This happens if your target is not configured to enable the secure shell server (sshd).  Try connecting to the target via the web interface or MAX to correct the setting."

Sufficient?

closed 2017-02-16 by pollockm

brbrindle, 2017-02-16:
yes that was the message... and yes that error will suffice

pollockm, 2017-02-16:
Great!  Fixed in Github, will be pushed out in the next update.  Thanks for the issue report - keep them coming!

commit f4fb489e (2017-02-16): Catch SSH disabled error

Fixes #281