LabVIEW-DCAF/TagEditorCore #284: Unable to remove module from config if plugin isnt available

brbrindle, 2017-02-16:
Should be able to delete modules fromconfiguration when plugin isn't available

jwgorman, 2017-08-08:
We could make this consistent with the LV project behavior - for example, show a scary missing glyph if you have an ethercat device but no ethercat driver.

Other considerations - if you save, you may lose the value of that module if you save it. We want to be explicit about that if we decide to go down that route.

dbendele, 2017-09-26:
when trying to load a configuration that references a plugin module that is not present on the system, the following error can be observed:

[image]

As we move to officially supported product, we want to make it easier for customers and developers to use the standard configuration editor to load and inspect configurations even when the configuration refers to 'missing' modules.

dbendele, 2017-09-28:
discussed with @becega 

recommendations: 
1. Extend Standard Editor.lvlib:StateCluster.ctl to have a new field "Read Only" that would be set to true if error on load. Customers would still be able to Save As in order to save modified configuration.
2. Update NISE_CEF_TreeSerializer.lvlib:strings to memory.vi to continue on error
3. For classes that cannot be found (currently returns error), instantiate a new placeholder object that retains serialized class data (class name, version, data) and can be inserted into system hierarchy tree
4. Create placeholder object that inherits from NISE_CEF_serializable configuration.lvclass that specifically implements To String, From String that preserves class data and has an associated glyph that identifies modules that could not be loaded.

rejected alternative:
NISE_CEF_TreeSerializer.lvlib:strings to memory.vi to continue on error and return error description that includes all missing modules but that would not allow the user to load configuration. This alternative was rejected because it did not sufficiently advance the workflow for the case of missing modules.

The recommended plan also allows for future workflow enhancements whereby a missing module may be removed/replaced/remapped.