LabVIEW-DCAF/TagEditorCore #286: Define Columns.vi pointing to missing VI

mcornwall, 2017-02-17:
Test Vis\editor node\Define Columns.vi 
depends on Set Column Width.vi that does not exist in the repository anymore.

mcornwall, 2017-02-21:
This may be related. The Test Node.lvproj was saved with some dependencies in vi.lib that are part of the Tag Editor Core package and therefor should have been pointing into the Dev environment. Uninstall tag editor core and open the dev version of Test Node.lvproj and you'll see what I'm talking about.

pollockm, 2017-02-21:
...this project is not used for anything these days, it looks like an old relic from table api development.  Is this bug holding you up in any way?  Would you be sad if we didn't fix it?

pollockm, 2017-02-21:
Or to put it another way, would a reasonable approach to fixing this be to exclude that directory from the VI Package build rather than making orphaned code happy again?

mcornwall, 2017-02-21:
My solution was to delete the offending code so that the package would build.
I have no problem if it disappears.

pollockm, 2017-02-21:
Removed the linkage to the missing VI.  The tageditorcore package already excludes the test vis directory from the package build, so this looks like it's only an issue for folks who are not using our VIPB.

commit fab2ecce (2017-02-21): Remove link to non-existent VI

Fixes #286
closed 2017-02-21 by pollockm