LabVIEW-DCAF/TagEditorCore #290: Add INI tokens for standard editor

pollockm, 2017-02-21:
We currently have the ability to right-click on the module editor node's subpanel and show block diagram.  This ability probably shouldn't be there for deployed applications - I'd like to put it behind an INI token that we document.

I suspect there are other features that make sense for a module developer but not for a user of the config editor.  It would be nice if such things could be selectable also, either via INI tokens or via a check of whether it is in dev or run time environment.

Beazurt, 2017-06-08:
Another example of this that we need a solution for is the scripting tools baked into the editor. Those won't work in a deployed application (assuming the editor is built as an EXE).

theSloopJohnB, 2017-07-24:
This could also be useful for #303 - we could store the last used config file in an ini file.

commit 396f0e51 (2018-10-19): ReOpen Configuration on Launch

Fix #303 and Fix #290

Added INI file to list of standard DCAF folders (along with cache and glyph folders) and created first token which lists the configuration open when the editor exits.
PR #395 (2018-10-22): ReOpen Configuration on Launch
Fix #303 and Fix #290

Added INI file to list of standard DCAF folders (along with cache and glyph folders) and created first token which lists the configuration open when the editor exits.
closed 2018-10-22 by becega