LabVIEW-DCAF/TagEditorCore #291: Deploy tool: CRIO targets file names with spaces do not transmit

brbrindle, 2017-02-28:
[image]

File names with spaces do not transmit with new SSH transfer methods, showing the above error message. Using old webdav method file names with spaces did work.

pollockm, 2017-02-28:
Suspected cause: the command line call needs to escape the file name with quotes.

No test case was done with a file containing a space.  Need to add one.

jwgorman, 2017-04-11:
Acceptance Criteria:
- Files with spaces deploy successfully
- Manually test

theSloopJohnB, 2017-06-06:
:( This is broken again. I need to make sure this is added to the manual deploy tests.

[image]

commit 7ab5fada (2017-07-10): Merge pull request #326 from LabVIEW-DCAF/deploy-tool-fixes

changed double quotes to single quotes for file name for proper escaping to fix #291
commit b835bed7 (2017-07-10): changed double quotes to single quotes for file name for proper escaping to fix #291
closed 2017-07-10 by becega