LabVIEW-DCAF/TagEditorCore #294: SSH deploy method breaks permissions preventing Labview from deploying rtexe

brbrindle, 2017-03-23:
When including .pcfg in deploy method from Labview development will fail with insufficient permissions if DCAF editor has ever been used to deploy .pcfg file.

problem is avoidable by not including it in RTEXE deployment, however could cause problems later.

pollockm, 2017-03-23:
We SSH files over using the admin account.   LabVIEW deploys files using the lvuser account.  lvuser cannot overwrite a file owned by admin.

Possible solution: set owner to lvuser on any .pcfg files after deploy as part of command line call to the SCP client.

jwgorman, 2017-04-13:
Acceptance Criteria
- Deploy first from deploy tool, deploy second from LabVIEW and the deploy works.
- Put in a test to deploy an admin owned file with LabVIEW's code path

PR #296 (2017-04-24): Deployment bugs Error Handling
Fixes #294, #189, #237, #291
commit 1327f597 (2017-04-24): Fix #294

Adding SCP Set Permissions VI to call chmod on remote target.
Putting Puts File.vi error handling into SubVI so set permissions can
use the same information.
Adding a testing VI stubbed out for integration tests (since you need
actual hardware)
closed 2017-04-24 by becega