LabVIEW-DCAF/TagEditorCore #310: TagEditorCore

dbendele, 2017-05-19:
Error on Deploy when multiple versions of LabVIEW present, but one of them does not have LabVIEW Real Time installed.
[image]

Beazurt, 2017-06-08:
Do we have a root cause for why this is happening? I can't think of any reason why it would. Does the deploy tool introduce LabVIEW RT as a dependency of DCAF?

pollockm, 2017-06-08:
...I would believe this could be caused if multiple LV versions are present, but the LV version in use did not have the syslog wrapper dependency installed.  I don't see how RT would affect, since the code uses the vi.lib constant and does not reference the RT directories.

Beazurt, 2017-06-08:
I just ran into this same issue. My dialog also showed that it was looking for the .so files in the LabVIEW 2015 folder which I no longer have installed (but there is still a legacy LabVIEW 2015 folder in program files that is mostly empty). I haven't investigated yet, but I would think the Deploy Tool should be looking for those files in the version of LabVIEW that is being used. In my case this was 2017.

Beazurt, 2017-06-08:
When I created a new config file in 2017 it worked fine. When I used the shipping example config, created in 2013, I got the error above.

I am pretty sure the problem is that we create the default paths for these files in the version of LabVIEW that created the configuration file. However since this only happens once, if you try to deploy a configuration file in a version of the editor running a different version of LabVIEW, it doesn't work.

theSloopJohnB, 2017-09-25:
We should discuss this to decide whether to fix or not.

closed 2017-10-24 by jbellamyNI