LabVIEW-DCAF/TagEditorCore #319: Some Datatypes can break the editor parser

mcornwall, 2017-06-02:
Specifically, I put two doubles in a cluster and made that a datamember of a modules configuration class.
When sending the module config 'to string' i used flatten to string and wired it into the included data wrapping vi.
Save the config to disk and then reopen it with the config editor.
 
It writes a character to file that causes the file to end at that point when parsing it back in. The result will be an error in the datatype read and anything in the config file beyond that point is lost to the system.

becega, 2017-06-02:
Are you working with the Static Module? Does the error happens in the from string inside he module or at a higher level?

mcornwall, 2017-06-02:
I believe that it has to do with the value held in the string.
the value flattened to a string looks like this:
<scale>@ย \u0000\u0000</scale>
3FF0 0000 0000 0000 
That character seems to be causing the problem.

I probed the string coming into 'Unflatten from String.vi' for the module an this is what it is getting. Notice that it ends abruptly at the scale.
`<tbm_moduleversion>1.0.0</tbm_moduleversion>
<tbm_moduledata><tbm_rootversion>1.0.0</tbm_rootversion>
<tbm_rootdata>
<instanceID>B1933E28-360D-472B-9A1B-77F0F85809FF</instanceID>
<kvpairs>
<SCFG_AliasName>RTX Modbus configuration</SCFG_AliasName>
<SCFG_ViewClass>RTX Modbus editor node.lvclass</SCFG_ViewClass>
</kvpairs>
</tbm_rootdata></tbm_moduledata>
<MBSlaveChannels><400002><ch_class>input data</ch_class><default></default><meta></meta><ch_type>Double</ch_type><ch_version>1.0.0</ch_version><mbc_mbtype>holding registers</mbc_mbtype><mbc_startaddr>1</mbc_startaddr><mbc_subindex>0</mbc_subindex><scale>?ð`

smithed, 2017-06-03:
If there is a flatten to/from json anywhere, it can't handle null characters (I believe it flattens 0x00 but can't parse \u0000 back out).

becega, 2017-06-05:
Its not a bug, will keep it open so we add this information to documentation, Its better to use string to number or Format into string. If you use flatten to string it might generate illegal characters that can have issues if not escaped correctly.

theSloopJohnB, 2017-06-27:
We should make sure the to and from string template VIs contain instructions for correct behavior.

theSloopJohnB, 2017-07-11:
Moving this issue to the templates repo.

closed 2017-07-11 by theSloopJohnB