LabVIEW-DCAF/TagEditorCore #320: Includes scripting path does not script the PL_Target RIO Runtime.lvclass

theSloopJohnB, 2017-06-06:
When attempting to use an FPGA bitfile I specified in the configuration file, I saw an error complaining about "PL_Target RIO Runtime.lvclass." 

I was able to resolve the issue by manually finding and adding that class to my static includes. It should have been automatically scripted based on my having checked "load bitfile."

theSloopJohnB, 2017-09-13:
Investigated some more. In CRIOUI.vi for generate includes file VI, we pass in the descendants of the the PLTarget cRIO.lvclass, but not itself!

[image]

I added an array insert to remedy the issue. When I re-ran my main.vi I no longer got the error about needing to load the runtime class! woot!

This seems like an issue that would surface for other target runtimes as well - I'm guessing the CVS one because it also has FPGA settings. Should I fix it there too?

DCAF-Builder, 2017-09-18:
Yes, we should fix this on CVS also.

PR #351 (2017-09-18): makes RIO node drop self and descendants
The issue has some more details and a screenshot of the fix. Basically, during init, the runtime is supposed to initialize the bitfile. That wasn't happening because the runtime wasn't being dropped as a part of the scripting section.

When you look at the scripting section, it drops child nodes, but not itself! The answer was to add itself to the array of things to drop.

fixes #320
commit 9f4d2e85 (2017-09-18): fix #320 by making RIO node drop self and descendants
closed 2017-09-18 by pollockm