LabVIEW-DCAF/TagEditorCore #330: Putty install as silent

brbrindle, 2017-07-14:
after a few installs of DCAF packages might recommend installing Putty silently.

silent install/uninstall via msiexec /q, as built into the msi. 

as currently exists you have to baby sit the install to make sure Putty installs before it installs the actual VI package.

as of right now the dialog to prompt the install of putty ends up behind Labview and VIPM, with no clear taskbar item for what happened and why the package hasnt install. with no dialog and silent install of putty we wouldnt have a problem. as of right now i believe SSH is the prefered deploy method for RT targets so why even make putty known... just install it and have a note in the package.

smithed, 2017-07-15:
I agree with the last point and suggested it a while back too...putty is completely standalone and doesn't need an installer. Just package up the putty.zip (a whopping 2 MB install from here https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) and include it. 

Or if you want to make it explicit, is there a legally acceptable way to make a separate package, put it up as hidden on the tools network, and make it a dependency for the dcaf packages? That way the associated license is part of the chunk of information all users completely ignore while installing software.

theSloopJohnB, 2017-10-10:
fixed by 
https://github.com/LabVIEW-DCAF/TagEditorCore/commit/5d364a5ab3ead235176befe84ed765d2ae8c6b24

closed 2017-10-10 by theSloopJohnB