LabVIEW-DCAF/TagEditorCore #333: Scripting module includes VI crashes labview 2017 with attached config

theSloopJohnB, 2017-07-18:
This config file: [test-config.zip](https://github.com/LabVIEW-DCAF/TagEditorCore/files/1157445/test-config.zip)

Does not behave well with LabVIEW 2017. I need to try it out with other versions.

If you try and script the module includes.vi, you get the following error:
[image]
Suspiciously, it mentions EIP in the reporter. I downloaded and attached the NIER report as well, in case it is helpful.
[error-report.zip](https://github.com/LabVIEW-DCAF/TagEditorCore/files/1157466/error-report.zip)

The configuration was created by making a PC target, one standard engine, and all possible modules I could put under it.

More investigation is required.

smithed, 2017-07-19:
EIP is the x86 instruction pointer register.

theSloopJohnB, 2017-07-19:
I did some more work on it this morning. 

I thought that maybe it was a trouble module, or a combination of modules that was the issue. I attempted to script by cutting out half of the modules - still had a crash. I thought I was on the right track. I then attempted to build a configuration file one module at a time, seeing which one being added produced a crash. This time, I had no crash! I tried deleting the contents of my includes VI in order to script it from a fresh slate - no crash. 

Then I tried to reproduce it with the original file attached to this issue. No crash this time!

I don't know why the crash stopped happening.

I want to see if anyone else can reproduce this and then decide what to do.

dbendele, 2017-07-19:
did not reproduce crash in LV 2014,
with attached configuration, scripting resulted in prompt to Find control named:
"ethernetip.lvlib:etherneti[.ctl" (which I canceled and ignored future prompts) 
Target Includes VI included all runtime objects except  Ethernet IP runtime

jwgorman, 2017-08-08:
Closing non-reproducible

closed 2017-08-08 by jwgorman