LabVIEW-DCAF/TagEditorCore #335: Deleting module does not reset module count

bkindinger, 2017-07-26:
If a module is only allowed a single instance, and you remove it, the parent node does not accurate count how many modules are added.

pollockm, 2017-09-15:
Marking as major bug.  If a user adds a scan engine module, then deletes it, they can never add another scan engine module to that target ever again.  This will be a support issue.

jacobson3, 2018-01-30:
It looks like the problem is coming from the RemoveChileFromNode.vi in Engine Parent Editor Node.lvclass. As seen below, the Targets.ConfigurationClassName keeps track of all modules which are used under that target. When an item is removed we search for it in the list of modules and delete it from the array. The problem is simply that the search 1d array is mis-wired into the length input and not the index.

[image]

commit bfa0e93a (2018-01-30): fixed #335

RemoveChildFromNode now correctly maintains list of modules in use by
target as discussed in #335

Also cleaned up and documented VI a bit.
commit 4b989165 (2018-01-30): Merge pull request #378 from LabVIEW-DCAF/Fix-Target-Module-Count

fixed #335
closed 2018-01-30 by becega