LabVIEW-DCAF/TagEditorCore #343: Resource reservation

pollockm, 2017-08-18:
We've seen a few cases where modules need to know some information about what other modules are doing to prevent collisions.

Examples:
Motion control on an axis - need to make sure two instantiations are not trying to control the same axis
DAQmx devices - we should not let the same resource be used in multiple engines/module

I'd like a way for modules to register for 'named' resources using a common nomenclature, and to have facilities for keeping track of which resources are used by which modules.

If a module tries to reserve a resource used by something else, I'd like it to fail the editor validation check and report back the engine and module name of the conflicting resource(s).

pollockm, 2017-08-18:
One relatively simple scheme is to have a "resources" variant, use key-value pairs to store the GUID/name/whatever of the module that first reserved that resource.

Need to have modules add reservations on configuration or creation, and also remove reservations on module deletion or config change.

Beazurt, 2017-08-22:
Tanner and I recently mocked up something similar for VeriStand, although it was specifically for FPGA resources to protect multiple custom devices that need to interact with the same bitfile. It would be interesting to see if we could use the same underlying LabVIEW API for both use-cases.

DCAF-Builder, 2017-09-16:
@Beazurt - Where might I find that implementation?