LabVIEW-DCAF/TagEditorCore #348: Where did our error definitions go?

pollockm, 2017-09-12:
We used to have good error descriptions for our error codes.  Now DCAF errors are showing up as undefined:

[image]

What happened to our error definition files?  

Marking this as prioritize, since this will be a support issue.

pollockm, 2017-09-13:
In at least the case screenshotted above, the error dialog happens here: 
[image]

pollockm, 2017-09-13:
I believe this is part of the template for Delete Line provided.

theSloopJohnB, 2017-09-13:
Checking through the error files, I found a file that used to define the error @pollockm found in this issue - but has not changed in 3 years. It seems like this is more likely something we missed.
https://github.com/LabVIEW-DCAF/ModuleInterface/commits/master/source/TagBusModuleFramework-errors.txt

I ran into a similar issue with a different error code and thought they may be related:

I got this while running a main.vi.
[image]

My error seems to come from the class discovery singleton. Concerningly - I do have [CEF-errors.txt](https://github.com/NISystemsEngineering/CEF/blob/master/Trunk/CEF-errors.txt) installed into C:\Program Files (x86)\National Instruments\LabVIEW 2014\project\errors.

The undefined error only happens when running on the RT target - not the host. I talked to @dbendele about errors and he mentioned that error definitions usually need to be installed to the target to see them.

Is it generally the case that errors we custom define in DCAF are not defined on the rt target, and that we doom our users to generic messages on RT? Do we have a decent way of deploying these files?

theSloopJohnB, 2017-09-14:
I verified that this is a part of the template for both static and dynamic modules.

theSloopJohnB, 2017-09-14:
Closing this - moved to issues https://github.com/LabVIEW-DCAF/ModuleInterface/issues/32 and https://github.com/LabVIEW-DCAF/TagEditorCore/issues/352

closed 2017-09-14 by theSloopJohnB

pollockm, 2017-09-14:
...and yet the scan engine issue happens on a desktop target.