LabVIEW-DCAF/TagEditorCore #364: Multiple instances of modules cause bleedover of config displayed

pollockm, 2017-10-13:
[image]

Ex: 
[dds.zip](https://github.com/LabVIEW-DCAF/TagEditorCore/files/1383335/dds.zip)

pollockm, 2017-10-13:
To reproduce: 
View the names of the modbus channels on each target.
change the name of the modbus channel under one target.
Click on the modbus module on the other target.  Note that the name given previously has now bled over here.

We're not correctly clearing the display when switching in the subpanel.  This caused massive confusion in the customer course I'm teaching this week.

pollockm, 2017-10-13:
This issue was reproduced with the modbus and DDS modules (same behavior).  Not reproduced in EIP Tags or EIP.

I assume this may be template-related.

becega, 2017-10-16:
Can reproduce it with the new versions it wasn't there at least until tag editor core 2.4.0.6. WHen updating core it also update 2 other modules, the updates in my machine were:
Tag Editor Core 2.4.0.6 --> 2.4.0.33
Module Interface 2.3.0.11 -->  2.3.0.17
Execution Interface 2.3.0.11 --> 2.3.0.7
Most Likely its in tag editor core

becega, 2017-10-16:
The issue was introduced on the build 0.32 that i think is related to this pull request:
https://github.com/LabVIEW-DCAF/TagEditorCore/pull/356

becega, 2017-10-17:
The bug was introduced in this VI \Tag Editor Core\Table API\Listbox\Init.vi

Value signaling causes the error and just switching value solves the issue and keeps the previous bug solved.

dbendele, 2017-10-18:
@becega , thanks for investigating! Are you going to make the change? 

As a side note, I don't yet understand why the value signaling would lead to this issue.

closed 2017-10-18 by dbendele